ARM-software/psa-arch-tests #230: Failing RSA Key Pair

zaya-mc, 2020-08-19:
Hi everyone, 

In test 39 and also all test which use the same key, RSA Key Pair tests fail. 

The key pair is like below; 

```
const uint8_t rsa_128_key_pair[] = {
0x30, 0x82, 0x02, 0x5e, 0x02, 0x01, 0x00, 0x02, 0x81, 0x81, 0x00, 0xaf, 0x05,
0x7d, 0x39, 0x6e, 0xe8, 0x4f, 0xb7, 0x5f, 0xdb, 0xb5, 0xc2, 0xb1, 0x3c, 0x7f,
... TRUNCATED ...
0x5e, 0x71, 0x78, 0x4e, 0x25, 0xac, 0xa4, 0xba, 0xf2, 0x5b, 0xe5, 0x73, 0x8a,
0xae, 0x59, 0xbb, 0xfe, 0x1c, 0x99, 0x77, 0x81, 0x44, 0x7a, 0x2b, 0x24};
```

Once we parse this key using mbedtls_pk_parse_key() we get following values
```

Field : N (len 128) 
0xc3, 0x6b, 0x5f, 0x98, 0xf7, 0x0e, 0x69, 0x3d, 0xa5, 0x4d, 0x83, 0xd6, 0xf7, 0x60, 0x86, 0x9e, 
0x86, 0x77, 0xf0, 0x33, 0x47, 0xfe, 0xbf, 0x2a, 0x3a, 0x26, 0xde, 0x4e, 0xd3, 0x9c, 0x2b, 0xb8, 
0x4c, 0x32, 0x9c, 0x81, 0xc1, 0xc2, 0x59, 0xf8, 0xcb, 0xa7, 0xd3, 0x94, 0xbd, 0xae, 0x96, 0xe7, 
0x02, 0xb2, 0x02, 0xcd, 0x48, 0x19, 0xc6, 0xc2, 0x06, 0xbc, 0x83, 0x44, 0xe5, 0xee, 0xdd, 0x49, 
0x23, 0xf1, 0x7a, 0x07, 0xb2, 0xee, 0x56, 0x89, 0x1d, 0x6e, 0x0e, 0x4c, 0xef, 0x24, 0x35, 0x44, 
0xd1, 0x20, 0xc1, 0xb6, 0xbb, 0x2d, 0x04, 0x28, 0x96, 0x12, 0xe1, 0x49, 0x12, 0x53, 0x11, 0x97, 
0xc7, 0x25, 0x2d, 0xb1, 0xb0, 0xfe, 0xe1, 0x1e, 0x54, 0x0b, 0x47, 0xa2, 0x8a, 0xaa, 0x54, 0xa6, 
0xe5, 0x7f, 0x3c, 0xb1, 0xc2, 0xb5, 0xdb, 0x5f, 0xb7, 0x4f, 0xe8, 0x6e, 0x39, 0x7d, 0x05, 0xaf, 

Field : E (len 4) 
0x01, 0x00, 0x01, 0x00, 
 
Field : P (len 64) 
0x13, 0x81, 0x48, 0x6b, 0xc6, 0x74, 0xa1, 0x71, 0x98, 0x2c, 0x2e, 0xb5, 0xd6, 0x12, 0x5c, 0xd4, 
0x7f, 0xd9, 0x60, 0xd1, 0xb9, 0x28, 0x19, 0xbe, 0xa5, 0x6c, 0x83, 0x07, 0xf6, 0x78, 0xbd, 0xad, 
0x98, 0xfd, 0x67, 0x94, 0x1a, 0x4e, 0xfe, 0xbb, 0x5c, 0x97, 0x1b, 0x7f, 0x40, 0x14, 0x22, 0xd8, 
0x81, 0x73, 0xeb, 0xe1, 0x2d, 0x22, 0x1d, 0xa8, 0x30, 0x9d, 0xbb, 0x41, 0x08, 0x76, 0xab, 0xe2, 

Field : Q (len 64) 
0x91, 0x70, 0x92, 0x65, 0x05, 0x84, 0x2e, 0x90, 0x74, 0x17, 0x11, 0x3f, 0xc7, 0xc3, 0x52, 0x86, 
0x33, 0xda, 0xab, 0x63, 0xb6, 0xce, 0x89, 0x17, 0xfc, 0xe7, 0x67, 0x64, 0xd3, 0x4f, 0xb4, 0x8b, 
0x0b, 0xa9, 0xf8, 0x9e, 0xa9, 0x90, 0xf9, 0xbb, 0xfc, 0xd5, 0x28, 0x9e, 0x68, 0x2e, 0x11, 0xdc, 
0x46, 0x1e, 0x85, 0xee, 0xc2, 0xc3, 0x20, 0x6f, 0x7d, 0xae, 0x59, 0x21, 0x60, 0x27, 0xab, 0xc5, 

```

Once we use this key in a specific secure element returns error this is because the P Q and N values are not consistent. 

 p * q should be equal to the modulus n, while p = ..0xe2, q =..0xc5 and n =..0xaf in the given example (the last digit of p*q is different of the last digit of n)

Are we missing something?

athoelke, 2020-08-19:
I think the fields in the parsed output are little-endian: the lower order bytes of P and Q do result in N:

    0x...8113 * 0x...7091 == 0x...6bc3

This may not explain the issue you are encountering though...

zaya-mc, 2020-08-20:
Hi @athoelke, yes you are right. It was about the endianness. So, we can close the ticket. Thank you.

closed 2020-08-20 by zaya-mc