ARM-software/psa-arch-tests #234: Build failure on msys2

ccli8, 2020-09-09:
My build environment is msys2/Windows 10. Following https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/sw_requirements.md and https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/dev_apis/README.md,, I set up my environment and build on msys2.

cmake (3.17.2) (windows native)
python (3.7.7) (windows native)
git (2.25.0) (windows native)
toolchain (GNU Tools for Arm Embedded Processors 9-2019-q4-major) (windows native)

But meet error:

```
# cmake ../ -G"Unix Makefiles" -DTARGET=tgt_dev_apis_tfm_an521 -DCPU_ARCH=armv8m_ml -DSUITE=CRYPTO -DPSA_INCLUDE_PATHS="`readlink -f ../../../trusted-firmware-m/interface/include`"
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error in CMakeLists.txt:
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error in CMakeLists.txt:
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/CMakeFiles/CMakeOutput.log".
See also "C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/CMakeFiles/CMakeError.log".
```

Do you have plan to enable build on msys2?

ccli8, 2020-09-14:
Installing msys2 host gcc toolchain and updating psa-arch-tests to master d6612f47c1a7d79d6b8c41dc5e50c0554f9be111, I can go further:

```sh
CCLi8@oanbe02023 MINGW64 /c/ccli8/iot/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD
# cmake ../ -G"Unix Makefiles" \
> -DTARGET=tgt_dev_apis_tfm_an521 \
> -DTOOLCHAIN=GNUARM \
> -DCPU_ARCH=armv8m_ml \
> -DSUITE=CRYPTO \
> -DPSA_INCLUDE_PATHS="`readlink -f ../../../trusted-firmware-m/interface/include`"

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: C:/msys64/usr/bin/cc.exe
-- Check for working C compiler: C:/msys64/usr/bin/cc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/msys64/usr/bin/c++.exe
-- Check for working CXX compiler: C:/msys64/usr/bin/c++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: C:/Python37/python3.exe (found suitable version "3.7.7", minimum required is "3")
-- Found Git: C:/Program Files/Git/mingw64/bin/git.exe (found version "2.25.0.windows.1")
-- [PSA] : ----------Process input arguments- start-------------
-- [PSA] : TARGET is set to tgt_dev_apis_tfm_an521
-- [PSA] : SUITE is set to CRYPTO
-- [PSA] : TOOLCHAIN is set to GNUARM
-- [PSA] : CPU_ARCH is set to armv8m_ml
-- [PSA] : Defaulting VERBOSE to 3
-- [PSA] : Defaulting INCLUDE_PANIC_TESTS to 0
-- [PSA] : Defaulting WATCHDOG_AVAILABLE to 1
-- [PSA] : Defaulting SP_HEAP_MEM_SUPP to 1
-- [PSA] : ----------Process input arguments- complete-------------
-- [PSA] : Creating testlist.txt 'available at C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/crypto_testlist.txt'
Non-secure test entry symbol list:
        C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/test_entry_list.inc,
        C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/test_entry_fn_declare_list.inc
-- [PSA] : C  compiler used 'C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major/bin/arm-none-eabi-gcc.exe'
-- [PSA] : ASM  compiler used 'C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major/bin/arm-none-eabi-gcc.exe'
-- Configuring done
-- Generating done
-- Build files have been written to: C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD

```

Then I meet duplicate `addr_t` typedefs error:

```sh
# cmake --build .
make[1]: Entering directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
make[2]: Entering directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
Scanning dependencies of target psa_generate_database_prerequisite
make[2]: Leaving directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
make[2]: Entering directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
make[2]: Leaving directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
[  0%] Built target psa_generate_database_prerequisite
make[2]: Entering directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
Scanning dependencies of target psa_generate_database
make[2]: Leaving directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
make[2]: Entering directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
[  1%] Creating directories for 'psa_generate_database'
[  1%] No download step for 'psa_generate_database'
[  2%] No patch step for 'psa_generate_database'
[  3%] No update step for 'psa_generate_database'
[  3%] Performing configure step for 'psa_generate_database'
-- Configuring done
[... 13 lines cut ...]
                 from C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/val/common/val_target.h:21,
                 from C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/targetConfigGen.c:2:
C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/platform/targets/common/nspe/pal_common.h:32:29: error: conflicting types for ‘addr_t’
   32 | typedef uint32_t            addr_t;
      |                             ^~~~~~
In file included from /usr/include/sys/types.h:224,
                 from /usr/include/stdio.h:61,
                 from C:/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/targetConfigGen.c:1:
/usr/include/machine/types.h:74:15: note: previous declaration of ‘addr_t’ was here
   74 | typedef char *addr_t;
      |               ^~~~~~
make[5]: *** [CMakeFiles/TargetConfigGen.dir/build.make:67: CMakeFiles/TargetConfigGen.dir/C_/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/targetConfigGen.c.obj] Error 1
make[5]: Leaving directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/src/psa_generate_database-build'
make[4]: *** [CMakeFiles/Makefile2:68: CMakeFiles/TargetConfigGen.dir/all] Error 2
make[4]: Leaving directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/src/psa_generate_database-build'
make[3]: *** [Makefile:130: all] Error 2
make[3]: Leaving directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD/src/psa_generate_database-build'
make[2]: *** [CMakeFiles/psa_generate_database.dir/build.make:75: src/psa_generate_database-stamp/psa_generate_database-install] Error 2
make[2]: Leaving directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/psa_generate_database.dir/all] Error 2
make[1]: Leaving directory '/c/CCLi8/IoT/mbed/mbed-tfm/psa-arch-tests/api-tests/BUILD'
make: *** [Makefile:84: all] Error 2
```

If I disable `addr_t` in my msys2 host gcc toolchain, I can complete the build process successfully. But in my opinion, it would be better that the `addr_t` in `nspe/pal_common.h` can be renamed to avoid such name collision.

kotegowder, 2020-09-15:
Hi,

At present we support Linux and Windows (mingw) build environments.
Msys support is not in our current roadmap.

But we shall revisit the requirement and come back with the details.

For now I would like to close these issues raised regarding msys. Let us know!