ARM-software/psa-arch-tests #239: Defect in QCBOR git clone destination folder check

davidhuziji, 2020-10-20:
Hi all,

A fatal error will occur if running CMake of PSA Arch test at a second time without cleaning up the build folder.
`fatal: destination path 'psa_qcbor' already exists and is not an empty directory.
CMake Error at <DIR>/api-tests/CMakeLists.txt:324 (message):
  git clone failed for https://github.com/laurencelundblade/QCBOR.git`

QCBOR folder does already exist since it is a second build. However, [line 318](https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/CMakeLists.txt#L318) fails to check whether the clone destination of QCBOR exists and therefore still trigger the git clone.

It looks like line 318 should check `${CMAKE_CURRENT_BINARY_DIR}/${PSA_TARGET_QCBOR}`, instead of `${PSA_TARGET_QCBOR}` alone, because [line 322](https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/CMakeLists.txt#L322) executes git clone under ``${CMAKE_CURRENT_BINARY_DIR}.

Any comment is welcome!

jk-arm, 2020-10-20:
the issue will be resolved in the upcoming release, till then please modify the check with ${CMAKE_CURRENT_BINARY_DIR}/${PSA_TARGET_QCBOR} at line number 318 as suggested above

jk-arm, 2020-11-17:
fixed

closed 2020-11-17 by jk-arm