ARM-software/psa-arch-tests #246: build PSA API TEST fail under win10

suewu2020, 2020-11-19:
SW version: TF-M commit ID:d49b514fdbd6e1367db8969e938619e3d615e617 
                     PSA test suite is download automatically during build process with below info
Scanning dependencies of target psa_arch_tests-populate
[ 11%] Creating directories for 'psa_arch_tests-populate'
[ 22%] Performing download step (git clone) for 'psa_arch_tests-populate'
Cloning into 'psa_arch_tests-src'...
remote: Enumerating objects: 794, done.
remote: Counting objects: 100% (794/794), done.
remote: Compressing objects: 100% (480/480), done.
remote: Total 9799 (delta 529), reused 498 (delta 305), pack-reused 9005
Receiving objects: 100% (9799/9799), 13.92 MiB | 6.52 MiB/s, done.
Resolving deltas: 100% (7182/7182), done.
Note: switching to '90c8e680'.

in win10 cmd, build the PSA API TEST  with below command:
cmake -S . -B cmake_build -G"Unix Makefiles" -DTFM_PLATFORM=mps2/an521 -DTFM_TOOLCHAIN_FILE=toolchain_ARMCLANG.cmake -DTEST_PSA_API=CRYPTO

and get the error info:
-- [PSA] : ----------Process input arguments- start-------------
CMake Error at cmake_build/lib/ext/psa_arch_tests-src/api-tests/CMakeLists.txt:108 (_check_arguments):
  Unknown CMake command "_check_arguments".

jk-arm, 2020-11-20:
hope the cmake version is >=3.15

suewu2020, 2020-11-20:
yes, cmake is 3.17

ccli8, 2020-12-01:
Also meet the same issue. Per my finding, trusted-firmware-m also has the path `common/Utils`. It can be fixed by changing:

https://github.com/ARM-software/psa-arch-tests/blob/608a082c56395fe579bcd0f2723b05904595329d/api-tests/CMakeLists.txt#L32

To:

```
include(${PSA_ROOT_DIR}/tools/cmake/common/Utils.cmake)
```

jk-arm, 2020-12-07:
we have never built the PSA arch test in a windows environment, maybe i can make this as a duplicate to this enhancement reported
https://github.com/ARM-software/psa-arch-tests/issues/234

jainvikas8, 2020-12-09:
I think this issue can be fixed with the above suggestion, as it creates problems for Windows 10 machine. 

Tried with using external repo
```
cmake -S . -B cmake_build -G"Unix Makefiles" -DTFM_PLATFORM=musca_b1 -DTFM_TOOLCHAIN_FILE=toolchain_GNUARM.cmake -DTFM_PSA_API=ON -DTEST_PSA_API=CRYPTO -DMBEDCRYPTO_PATH=../mbedtls -DPSA_ARCH_TESTS_PATH=../psa-arch-tests -DMCUBOOT_PATH=../mcuboot -DTFM_TEST_REPO_PATH=../tf-m-tests
```

LDong-Arm, 2020-12-16:
Same issue on macOS.

LDong-Arm, 2021-01-13:
The reason seems to be that on Windows or macOS, the slash `/` in `include("common/Utils")` cannot be resolved as "subdirectory". It's an OS-dependent behaviour. So, if you move `common` to be part of `CMAKE_MODULE_PATH` and `include("Utils")` instead, it works fine on all systems.

But @ccli8's suggestion to use full include path is preferred, for consistency with the rest of that CMakeLists.txt. Could make a PR if you prefer. @jk-arm

jk-arm, 2021-01-13:
sure, we will look and revert back. thankyou

jk-arm, 2021-01-13:
please do make the PR, we will review the PR

PR #250 (2021-01-18): api-tests: fix CMakeLists include on non-Linux
Fixes: #246

Credits to @ccli8(ccli8@nuvoton.com) for finding the fix.

When we use CMake's `CMAKE_MODULE_PATH` feature, the module name being included should not contain slashes (`/`) which can't be resolved (from regular file names) by CMake on Windows and macOS hosts, resulting in failure of `include("common/Utils")` on those systems.

Since other modules are included with full paths which work in all cases, do the same for `common/Utils.cmake`.

@jk-arm Please review, thanks.
commit 9d4e7df5 (2021-01-18): api-tests: fix CMakeLists include on non-Linux

Fixes: #246

Credits to @ccli8(ccli8@nuvoton.com) for finding the fix.

When we use CMake's CMAKE_MODULE_PATH feature, the module name
being included should not contain slashes ("/") which can't be
resolved by CMake on Windows and macOS hosts, resulting in
failure of `include("common/Utils")` on those systems.

Since other modules are included with full paths which work in
all cases, do the same for common/Utils.cmake.
closed 2021-01-18 by jk-arm