ARM-software/psa-arch-tests #247: Support Firmware Framework test for MUSCA S1 target (IPC)

jainvikas8, 2020-12-10:
Please add support of **IPC** test suite for ARM_MUSCA_S1 target. This target is supported in Mbed OS and TF-M.

evedon, 2020-12-24:
Could you let us know when you can work on this?

jk-arm, 2021-01-04:
https://github.com/ARM-software/psa-arch-tests/tree/master/api-tests/platform/targets/tgt_dev_apis_tfm_musca_s1, the s1 platform is suppored already

jainvikas8, 2021-01-04:
We are taling about IPC - firmware framework - The target goes by **tgt_ff_tfm_musca_s1** ?

jk-arm, 2021-01-04:
in CMKAE command, please mention the target as tgt_ff_tfm_musca_s1, the musca_s1 will be picked up

jainvikas8, 2021-01-04:
Is this documented somewhere? If `tgt_ff_tfm_musca_s1` and `tgt_ff_tfm_musca_b1` uses the same alias..

jainvikas8, 2021-01-04:
> in CMKAE command, please mention the target as tgt_ff_tfm_musca_s1, the musca_s1 will be picked up

Target not supported when using `tgt_ff_tfm_musca_s1` on https://github.com/ARM-software/psa-arch-tests/commit/13c6e2c439c5122b9f0e414ad6a687af809975fd
```
 -- [PSA] : ----------Process input arguments- start-------------..
 CMake Error at CMakeLists.txt:144 (message):..
 [PSA] : Error: Unsupported value for -DTARGET=tgt_ff_tfm_musca_s1,..
supported targets are :..
common;tgt_dev_apis_stdc;tgt_dev_apis_tfm_an521;tgt_dev_apis_tfm_an524;tgt_dev_apis_tfm_an539;tgt_dev_apis_tfm_musca_a;tgt_dev_apis_tfm_musca_b1;tgt_dev_apis_tfm_musca_s1;tgt_dev_apis_tfm_nrf;tgt_dev_apis_tfm_psoc64;tgt_ff_tfm_an521;tgt_ff_tfm_musca_a;tgt_ff_tfm_musca_b1..
```

jk-arm, 2021-01-04:
1. if you are compiling the standalone psa arch test for musca_s1 then the target should be "tgt_dev_apis_tfm_musca_s1"
2. if you are trying to compile through the TFM then TFM readme mentioned target name should be used, 

for 1st we can help you and the 2nd you should contact TFM team for proper alias mapping,

give us some time, we will try to reproduce and solve if required

jk-arm, 2021-01-06:
i am able to compile the build for musca_s1, if you are compiling in trusted firmware environment please follow the trusted firmware readme: 
https://ci-builds.trustedfirmware.org/static-files/ZlFG20BBiImhpv8EIHIISEnTVkrBD1kQUnVvX676rSYxNjA5OTIzMjgyMDc4Ojk6YW5vbnltb3VzOmpvYi90Zi1tLWJ1aWxkLWRvY3MtbmlnaHRseS9sYXN0U3RhYmxlQnVpbGQvYXJ0aWZhY3Q=/trusted-firmware-m/build/docs/user_guide/html/docs/getting_started/tfm_build_instruction.html

if you are compiling from psa arch test, please follow the readme file from psa arch test.

jainvikas8, 2021-01-06:
So to run the Firmware Framework test suite - IPC for ARM_MUSCA_S1 - we need to use - `tgt_dev_apis_tfm_musca_s1`?

jk-arm, 2021-01-06:
please use the link for running the IPC testcases
https://ci-builds.trustedfirmware.org/static-files/ZlFG20BBiImhpv8EIHIISEnTVkrBD1kQUnVvX676rSYxNjA5OTIzMjgyMDc4Ojk6YW5vbnltb3VzOmpvYi90Zi1tLWJ1aWxkLWRvY3MtbmlnaHRseS9sYXN0U3RhYmxlQnVpbGQvYXJ0aWZhY3Q=/trusted-firmware-m/build/docs/user_guide/html/docs/getting_started/tfm_build_instruction.html#build-for-psa-ff-ipc-compliance-tests

jk-arm, 2021-01-06:
use the macro -DTFM_PLATFORM=musca_s1

jk-arm, 2021-01-18:
internal jira to track the same with TFM team
https://jira.arm.com/browse/IOTPSW-3652

jk-arm, 2021-04-16:
the support added,

closed 2021-04-16 by jk-arm