ARM-software/psa-arch-tests #252: Option to skip test cases that are known to fail

LDong-Arm, 2021-02-10:
A number of test cases in the Crypto suite are known to fail and cause system resets. While the test framework is often able to pick up where it left off after the reset, an option to skip known failures would be ideal
* It gives users an indication everything that should pass does indeed pass
* It makes it easier/more feasible to integrate with any test infrastructures (e.g. the one we use for Mbed OS), as reboots are relatively hard to handle and lead to failures.

jk-arm, 2021-07-19:
@LDong-Arm , we are considering this as an enhancement and trying to bring the same to our roadmap.

jk-arm, 2021-08-05:
this is implemented and tracked through internal jira https://jira.arm.com/browse/PJ03206-417

jk-arm, 2021-08-05:
the resolution will be available in github on next release.