ARM-software/psa-arch-tests #253: Test problem with Zephyr OS

akiannillo, 2021-02-10:
Hi all,
I'm running the PSA functional tests (PROTECTED_STORAGE) with ZephyrOS and trustedfirmware-m (TF-Mv1.2.0) on qemu.
Unfortunately, it seems there are some problems with test 403 on the second round of setting: it simply fails (without messages) and restart the whole device.

```
[OTHER TEST BEFORE]
TEST: 403 | DESCRIPTION: Insufficient space check
                Input id is 01030000
                Input id is 01030000
                Input id is 02020000
                Input id is 02020000
[Info] Executing tests from non-secure
 
[Info] Executing PS tests
[Check 1] Overload storage space
Setting 0x00000200 bytes for UID 5
Setting 0x00000200 bytes for UID 6
Setting 0x00000200 bytes for UID 7
Setting 0x00000200 bytes for UID 8
Setting 0x00000200 bytes for UID 9
Setting 0x00000200 bytes for UID 10
Setting 0x00000200 bytes for UID 11
Setting 0x00000200 bytes for UID 12
Setting 0x00000200 bytes for UID 13
UID 13 set failed due to insufficient space
Remove all registered UIDs
Removing UID 5
Removing UID 6
Removing UID 7
Removing UID 8
Removing UID 9
Removing UID 10
Removing UID 11
Removing UID 12
[Check 2] Overload storage again to verify all previous UID removed
Setting 0x00000200 bytes for UID 5
Setting 0x00000200 bytes for UID 6
Setting 0x00000200 bytes for UID 7
Setting 0x00000200 bytes for UID 8
Setting 0x00000200 bytes for UID 9
[INF] Starting bootloader
[INF] Swap type: none
[INF] Swap type: none
[INF] Bootloader chainload address offset: 0x80000
[INF] Jumping to the first image slot
[Sec Thread] Secure image initializing!
Booting TFM v1.2.0
[Crypto] MBEDTLS_TEST_NULL_ENTROPY is not suitable for production!
*** Booting Zephyr OS build v2.5.0-rc1-209-g30634334a811  ***
[AGAIN FROM THE FIRST TEST AND RESTART AT THE VERY SAME POINT]
```

Any idea on how to explain/debug this?

jk-arm, 2021-02-18:
please increase the watchdog timer and try

akiannillo, 2021-02-18:
I just did it and worked setting all the timeout to 60 seconds.
Can I have an insight into why is that?

jk-arm, 2021-02-18:
not much from tech point of it, it could be external storage usage, we are trying to allocate the memory till it run out of space.

akiannillo, 2021-02-18:
ok, having almost the same problem with crypto. Is there a way to disable watchdog at all? I think it's because I'm running it in an emulator, can it be?

jk-arm, 2021-02-18:
please try using the function 
wd_timer_disable

akiannillo, 2021-02-19:
Not working.
I invoked `val_wd_timer_disable()` before `val_entry()`, and it returns 0 so I suppose it worked.
The Crypto test suite always stops and restarts at test 216/check4

jk-arm, 2021-02-22:
https://github.com/ARM-software/psa-arch-tests/tree/master/api-tests/dev_apis, try the suggested watch dog disable/enablemethod

jk-arm, 2021-04-16:
@akiannillo can we close this ticket?

akiannillo, 2021-06-02:
yes, there was a merge about this with PSA test.

closed 2021-06-02 by jk-arm