ARM-software/psa-arch-tests #254: Crypto tests require deprecated types and functions

Einhornhool, 2021-02-19:
Hi all =)
I am trying to build the PSA Crypto tests and I am getting some unknown type and implicit declaration errors from `api-tests/platform/targets/common/nspe/crypto/pal_crypto_intf.c`.

I checked it and according to the "Changes between 1.0 beta 3 and 1.0.0" section of the PSA Crypto specification, the `psa_key_handle_t` type and the `psa_open_key` and `psa_close_key` functions have been removed.
The functions `psa_asymmetric_sign` and `psa_asymmetric_verify` have been renamed to `psa_sign_hash` and `psa_verify_hash`. 

Apparently the tests still use the old types and functions, that I didn't implement, as they are not part of the specification anymore. Is there a configuration option I missed to specify which API version to use? Or are you planning on updating the code?

jk-arm, 2021-03-04:
The current psa arch test supports only the 1.0 beta3

Einhornhool, 2021-03-04:
Thank you for the answer. Are you planning on updating it any time soon?

jk-arm, 2021-03-04:
we are doing the development at present, we will update you Soon, our reference platform doest not support crypto 1.0 yet.

jk-arm, 2021-07-15:
could you please try the tag : v21.07_API1.2_ADAC_ALPHA and share your comments

jk-arm, 2021-07-21:
considering this for closure, please verify against the latest tag and come back

closed 2021-07-21 by jk-arm

Einhornhool, 2021-07-21:
Hey, I'm sorry I didn't get back to you sooner, I'm currently working on something else. But thank you for notifying me, I will try this as soon as I can.

jk-arm, 2021-07-21:
thankyou

Regards,
JK

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.