ARM-software/psa-arch-tests #261: MMIO regions in PSA FF tests undocumented

andvib, 2021-03-18:
The target configuration used in PSA FF tests set up three MMIO regions to be used in the tests:

- nspe_mmio
- server_partition_mmio
- driver_partition_mmio

These regions aren't documented in any way, and their purpose isn't clear from the code

Questions:
#1 What are the regions used for, and where?
#2 Are these memory regions expected to be in secure or non-secure RAM locations?

andvib, 2021-03-24:
@jk-arm Are you able to provide some input on this?

jk-arm, 2021-03-24:
@andvib, we will get back to you on this.

oyvindronningstad, 2021-04-15:
@jk-arm Could you provide input here?

jk-arm, 2021-04-15:
@jaypit02, could you please comment on the doubts?

jaypit02, 2021-04-20:
PFA FF tests needs these mmio regions address for below reasons:

- To check "mmio_regions" manifest field implementation against the spec rules such as defined region must belong to Secure Partition who specifies it and it must has exclusive access to an MMIO region. Secure Partitions are not permitted to share MMIO regions with other Secure Partitions.
- For tests who performs the invalid input buffer check for the IPC APIs. One of the example of such rule(test_i050)is, the call to psa_call() is PROGRAMMER ERROR if psa_outvec.base address is invalid for client. More detail can be found in testlist document - https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/psa_ipc_testlist.md

Attributes for regions should be as per below:
nspe_mmio - RW, NS_RAM
server_partition_mmio - RW, S_RAM
driver_partition_mmio - RW, S_RAM

Also the addresses of the regions should be specified such that it must follow the below condition. 
_nspe_mmio.start < server_partition_mmio.start < driver_partition_mmio.start

This helps us to cover corner cases for some of the rules for IPC APIs where base address of the input buffer is valid but end_address(base+size) of the buffer is invalid. In such case, IPC API must fail. See test_i052.

andvib, 2021-04-30:
This helped clearing things up so we were able to progress with the porting, thanks a lot!

closed 2021-04-30 by andvib