ARM-software/psa-arch-tests #262: FF test case 5 - server_test_strict_policy_higher_version calls process_connect_request but failed to get the expected signal

kevin-peng-hao, 2021-03-26:
This test is to check the behaviour of the framework when a client tries to request a service with a higher version.
The expected behaviour is that the client would be refused when psa_connect.
In TF-M, this is done in the SPM. So the request with a higher version would never get to service.

So calling process_connect_request to psa_wait the test signal would not get the expected signal.
The partition is only woken up when clients call it with another service request.
In this particular test case, it's the val_get_secure_test_result.
The signal sent is something about TEST_RETURN_RESULT while the partition is waiting for SERVER_STRICT_VERSION_SIGNAL.

Testing on IPC isolation Level 2 with GCC compiler fails this case.

jk-arm, 2021-04-16:
do you still face this issue? or can we close this one?

kevin-peng-hao, 2021-04-19:
This is newly found during TF-M 1.3 release.
Do you have changes (patches) for this issue?

mingyangSunshine, 2021-06-22:
test 048 also receives an unexpected signal after stateless rot change in FF-test.
detail:
in test 48, the test service is blocked waiting for a signal SERVER_UNSPECIFED_VERSION_SIGNAL , but the next signal sent from client is from status = val_get_secure_test_result(&handle);  in val_execute_non_secure_tests()
So the program receives a signal it does not expect. Thus test result shows "psa_wait returned with invalid signal value"

jk-arm, 2021-07-22:
could you please try on the latest label?

kevin-peng-hao, 2021-07-23:
This issue was found by armclang and gcc having different results for FF tests.
I tested on v21.07_API1.2_ADAC_ALPHA and they now have the same results.
So I think this issue has been fixed.

jk-arm, 2021-07-23:
Please close it. Thankyou for the confirmation

Regards,
JK

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

closed 2021-07-23 by kevin-peng-hao