ARM-software/psa-arch-tests #263: some questions about test_data58(psa_aead_update) and test_data59(psa_aead_finish)

FLAYhhh, 2021-04-19:
The prototype of psa_aead_update is:
```
psa_status_t psa_aead_update(psa_aead_operation_t *operation,
                             const uint8_t *input,
                             size_t input_length,
                             uint8_t *output,
                             size_t output_size,
                             size_t *output_length);
```
when alg is gcm, input_length = 28(as in test_data58), i think output_size will be 28 too, but why does it's actually expected to be 44 in test_data58?

I have same question about expected length and input length in psa_aead_finish too.

jk-arm, 2021-04-28:
psa aead finish is not implemented fully, we yet to test it from our side.

jk-arm, 2021-07-15:
it's not necessarily the input and out length have to be the same, its depends on the AEAD ALG and ALG TYPE and then input length. please read the "psa_aead_update".
I would suggest you to wait till the AEAD support is enabled at the architecture test suite side. be watchful for the upcoming releases. spec sniffed as below

[image]

jk-arm, 2021-07-16:
also some of the steps deliberately checking the length overflow.

jk-arm, 2021-07-21:
please do your testing on the latest released tag : v21.07_API1.2_ADAC_ALPHA,