ARM-software/psa-arch-tests #265: build: api-tests: cmake falls back to default host compiler

microbuilder, 2021-05-07:
When building `psa_arch_tests` in Zephyr with TF-M, cmake falls back to the default C compiler, rather than forwarding the correct ARM cross-compiler, resulting in build failure.

The use of the host compiler is caused by this: https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/CMakeLists.txt#L415-L436

Where the compiler is not passed on, but a C compiler is being looked up here:
https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/tools/scripts/target_cfg/CMakeLists.txt#L29

```cmake
# Let the CMake look for C compiler
project(TargetConfigGen LANGUAGES C)
```

**To Reproduce**

> The `twister` command is part of Zephyr, but you can see the clang compiler being called as the default with this test suite:

```bash
$ ./scripts/twister -p mps2_an521_nonsecure -N --inline-logs \
  -s samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test
```

This will result in trying to run the system C compiler (`/Library/Developer/CommandLineTools/usr/bin/cc` here, on OS X) instead of the toolchain that should be passed in (`arm-none-eabi-gcc`, `arm-zephyr-eabi-gcc`, etc.):

```
[2/5] Building C object CMakeFiles/TargetConfigGen.dir/Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c.o
FAILED: CMakeFiles/TargetConfigGen.dir/Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c.o 
/Library/Developer/CommandLineTools/usr/bin/cc  -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/val/nspe -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/val/common -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/platform/targets/common/nspe -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/platform/targets/common/nspe/crypto -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/platform/targets/tgt_dev_apis_tfm_an521/nspe -arch arm64 -o CMakeFiles/TargetConfigGen.dir/Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c.o -c /Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c
/Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c:1:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
```

**Expected behavior**

The toolchain triplet (`CROSS_COMPILE`) should be passed in to the cmake `ExternalProject` when running the PSA tests so that the right compiler is called.

jk-arm, 2021-06-30:
we will revisit this topic after july 11th as planned release activity psa arch test v1.2 on going

closed 2021-07-14 by prasanth-pulla

jk-arm, 2021-07-15:
> Closed #265<https://github.com/ARM-software/psa-arch-tests/issues/265> via #268<https://github.com/ARM-software/psa-arch-tests/pull/268>.
>
>
> IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

Currently working on unifying this requirement, TS team also had similar ask. We will communicate the final outcome.

https://review.trustedfirmware.org/c/TS/trusted-services/+/10561/1

Regards,
JK