ARM-software/psa-arch-tests #266: Request to shrink the `type` range test into INT16_MAX

KenLSoft, 2021-05-11:
Hi,

The test case `client_test_psa_call_with_allowed_type_values`, assigned the `INT32_MAX` as the maximum type number, this test case fails on the latest master branch of TF-M.

The reason is that TF-M now encodes the `type` as an `int16`, to combine this type with `inlen` and `outlen` makes the type a 32bit composite value. The reason for changing this is because:

FF-M 1.0 proposes 6 parameters for `psa_call` and the secure ABI support at most 4 parameters. Then a packing needs to be involved. To keep the type as 32-bit we have to encode it into a composite data structure, which costs extra overheads (extra memory check e.g.) inside the SPE. As TFM is trying to improve the performance now, this place is an obvious one to be refined. A rational solution is to shrink the `type` range, then combine it with `inlen` and `outlen`, which could save the overheads.

We have proposed a specification update request to the specification team, meanwhile, we changed the packing mechanism as an experiment to see if real user applications would report issues in practical usage --- till now, no application reported the shortage of `type` values except `arch-test`.

We know what the existing test case doing follows the existing specification, changing the maximum `type` value makes this test item not covering 100% range, but it is the most efficient way to help the public experiment inside implementation which would generate an output to the next iteration of specification.

We would provide a patch to update this part. Please help to check.

Thanks.

KenLSoft, 2021-05-14:
Thanks. Close the issue now.

closed 2021-05-14 by KenLSoft

LDong-Arm, 2021-05-14:
Thanks for the fix. Please update the revision fetched by trusted-firmware-m.