ARM-software/psa-arch-tests #273: PSA FF tests halted in MemManageFault

andvib, 2021-06-21:
A number of the Firmware Framework tests (i68-i73, i75-i77, i79) results in a MemManageFault. I'm running the tests together with TF-M, which by default handles these faults with an infinite loop. I understand that this behaviour is as expected, but the infinite loop causes the tests to stop execution until a manual reset is performed. This causes some problems when automating these tests to run on CI.

Is there a way to execute these tests, while at the same time avoiding that the MemManageFault-handler is called and enters the infinite loop?

jk-arm, 2021-06-30:
Hi @andvib , currently we are busy with the release activity, we will revisit this topic after July 10th.

jk-arm, 2021-07-19:
@andvib could you please try again with the latest tag "v21.07_API1.2_ADAC_ALPHA" ?

jk-arm, 2021-08-04:
we will consider this for closure

closed 2021-08-04 by jk-arm

andvib, 2021-08-12:
Sorry for late reply!

I tested with the tag and latest master, and I'm still seeing the same behaviour.

jk-arm, 2021-08-12:
@andvib , thank for the update,
we had tried in the local setup and those testcases were passing on the label, I think it's good to revisit again. 
could you please reopen this or recreate a new ticket?