ARM-software/psa-arch-tests #277: Test 46 and 47 give wrong expected result.

Summer-ARM, 2021-07-22:
In PSA Crypto spec 1.0, it defines:
psa_mac_compute():
 - PSA_ERROR_NOT_PERMITTED       The key does not have the PSA_KEY_USAGE_SIGN_MESSAGE flag, or it does
not permit the requested algorithm.
 - PSA_ERROR_INVALID_ARGUMENT         key is not compatible with alg .

So, the expected_status in 46:
- Test psa_mac_compute - Incompactible HMAC for CMAC   should **PSA_ERROR_INVALID_ARGUMENT**, not PSA_ERROR_NOT_SUPPORTED
- Test psa_mac_compute - Invalid usage    should be **PSA_ERROR_NOT_PERMITTED**, not PSA_ERROR_NOT_SUPPORTED

psa_mac_verify():
- PSA_ERROR_INVALID_ARGUMENT key is not compatible with alg .
- PSA_ERROR_NOT_PERMITTED The key does not have the PSA_KEY_USAGE_VERIFY_MESSAGE flag, or it
does not permit the requested algorithm.

So, the expected_status in 47:
- Test psa_mac_verify - Incompactible HMAC for CMAC  should be **PSA_ERROR_INVALID_ARGUMENT**, not PSA_ERROR_NOT_SUPPORTED
- Test psa_mac_verify - Invalid usage should **PSA_ERROR_NOT_PERMITTED**, not PSA_ERROR_NOT_SUPPORTED

jk-arm, 2021-07-22:
@Summer-ARM , this has been identified already with https://github.com/ARM-software/psa-arch-tests/issues/213
I will have to push the changes

jk-arm, 2021-07-22:
https://github.com/ARM-software/psa-arch-tests/commit/dfb944c8e9e3874d3958a456c5d1299646d99b50

Summer-ARM, 2021-07-22:
Hi JK,
The fix is not cover all the issues I mentioned.
Did not fix 'Test psa_mac_compute - Incompactible HMAC for CMAC' and 'Test psa_mac_verify - Incompactible HMAC for CMAC'.

Please double-check it.

jk-arm, 2021-07-22:
let me reopen this because this testcase input needs to be reworked.

closed 2021-07-22 by jk-arm