ARM-software/psa-arch-tests #283: Test 48 for CBC_NO_PADDING expects output length of 0, should be output length

Einhornhool, 2021-08-02:
According to the PSA Crypto specification, the parameter output_length of psa_cipher_encrypt() contains "On success, the number of bytes that make up the output". 
Test 48 Check 1 expects zero, but shouldn't it expect 32? Or did I misunderstand something?

jk-arm, 2021-08-03:
the test case logic needs to be reworked based on the outcome for https://github.com/ARM-software/psa-arch-tests/issues/215
please wait for it.

jk-arm, 2021-08-04:
for testcase 48, the fixes you can see here https://github.com/ARM-software/psa-arch-tests/blob/key_derivation

Kpakizeh, 2021-08-08:
> for testcase 48, the fixes you can see here https://github.com/ARM-software/psa-arch-tests/blob/key_derivation

The link is not working.

Einhornhool, 2021-08-08:
@Kpakizeh It didn't work for me either, but you can find it by checking out the branch called key_derivation.

Kpakizeh, 2021-08-08:
@Einhornhool  Thanks !.

jk-arm, 2021-08-08:
yes please do checkout.....

jk-arm, 2021-08-09:
fixes can be seen in branch : https://github.com/ARM-software/psa-arch-tests/tree/key_derivation

Kpakizeh, 2021-08-11:
@jk-arm  Regarding the fix mentioned here [test_c048](https://github.com/ARM-software/psa-arch-tests/blob/key_derivation/api-tests/dev_apis/crypto/test_c048/test_c048.c#L73), I do not see how it is correct to use the multi-step operation that takes an psa_cipher_operation_t  struct and set the IV, then have the same IV be used for [psa_cipher_encrypt](https://armmbed.github.io/mbed-crypto/html/api/ops/ciphers.html#c.psa_cipher_encrypt) which does not take that psa_cipher_operation_t object as part of its input ?. 
Can you point me to an implementation for psa_cipher_encrypt that accounts or tell me here what the correct use case looks like ?.

jk-arm, 2021-08-11:
@Kpakizeh , the psa_cipher_encrypt_setup needs to be removed as it testing only a one-shot cipher.
about the psa_cipher_encrypt implementation, you can look at the mbedtls(https://github.com/ARMmbed/mbedtls) repo.