ARM-software/psa-arch-tests #285: Inconsistency In Usage For psa mac multi-step/single-step operations

Kpakizeh, 2021-08-04:
There is consistency in the key usage flags for the keys between the multi-step and single step operations for all the PSA_MAC operations when the key type is _**PSA_KEY_TYPE_HMAC**_. 
Example:

- In [test_c046](https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/dev_apis/crypto/test_c046/test_data.h#L139) which is psa_mac_compute (single step operation) we see that the usage flag is **_PSA_KEY_USAGE_SIGN_HASH_**. 

- In [test_c027]( https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/dev_apis/crypto/test_c027/test_data.h#L40)  we see the key usage flag is set to **_PSA_KEY_USAGE_SIGN_MESSAGE_**. 

- It not correct to have two different usage flags since the operation is completing the same task. I believe the correct flag to use for both multi-step and single-step operations for the _**HMAC**_ algorithm  is **_PSA_KEY_USAGE_SIGN_MESSAGE_**.

jk-arm, 2021-08-11:
@Kpakizeh , please refer the below spec reference,

PSA_KEY_USAGE_SIGN_HASH (macro)
Permission to sign a message hash with the key.
#define PSA_KEY_USAGE_SIGN_HASH ((psa_key_usage_t)0x00001000)

This ag allows the key to be used to sign a message hash as part of an asymmetric signature operation, if
otherwise permitted by the key’s type and policy. The ag must be present on keys used when calling
psa_sign_hash().
This ag automatically sets PSA_KEY_USAGE_SIGN_MESSAGE: if an application sets the ag
PSA_KEY_USAGE_SIGN_HASH when creating a key, then the key always has the permissions conveyed by
PSA_KEY_USAGE_SIGN_MESSAGE, and the ag PSA_KEY_USAGE_SIGN_MESSAGE will also be present when the
application queries the usage ags of the key.
For a key pair, this concerns the private key.

jk-arm, 2021-08-16:
@Kpakizeh , hope you are agreeing on the explanation. if so lets close this ticket. i will wait for your reply.

jk-arm, 2021-08-16:
closing this ticket.