ARM-software/psa-arch-tests #289: PSA FF tests halted in MemManageFault

andvib, 2021-08-12:
A number of the Firmware Framework tests (i68-i73, i75-i77, i79) results in a MemManageFault. I'm running the tests together with TF-M, which by default handles these faults with an infinite loop. I understand that this behaviour is as expected, but the infinite loop causes the tests to stop execution until a manual reset is performed. This causes some problems when automating these tests to run on CI.

Is there a way to execute these tests, while at the same time avoiding that the MemManageFault-handler is called and enters the infinite loop?

Tested on latest master (e005834)

jk-arm, 2021-08-12:
thankyou we will reinvestigate this.

jaypit02, 2021-08-20:
Hi @andvib, 

These tests require system to get reset to be able to execute next test. One of following can be useful in your case:
- Test framework can recover from such infinite loop if secure watchdog is assigned to the Test Driver Partition. 
-  Or MemManageFault handler can be updated to call tfm_spm_hal_system_reset() to perform system reset.

andvib, 2021-08-25:
Thanks @jaypit02! Those solutions should work, at least now I know that this is supposed to happen.

Closing issue.

closed 2021-08-25 by andvib