ARM-software/psa-arch-tests #305: A compiler switch to enable or disable TLS12 PRF Key Derivation Tests

zaya-mc, 2022-01-28:
Can we have a Compiler Switch for TLS12 PRF Key Derivation Tests? It could be more flexible for custom target that does not implement it.
There are tests In the crypto test 20.

zaya-mc, 2022-01-28:
We can create a PR if wanted.

jk-arm, 2022-01-29:
Please create a PR, we will have a look at it, if possible we will bring along with this release

Regards,
JK

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

jk-arm, 2022-04-04:
closing this as https://github.com/ARM-software/psa-arch-tests/pull/313 merged

closed 2022-04-04 by jk-arm