ARM-software/psa-arch-tests #307: Crypto tests 224-225 "unsupported algorithm" check return code is too constrained

adeaarm, 2022-02-15:
For the Crypto test suite, tests 224 and 225 on the "unsupported algorithm" check expect the error code to be exclusively PSA_ERROR_NOT_SUPPORTED. By looking at this https://github.com/ARMmbed/mbedtls/pull/5272 it seems that both PSA_ERROR_NOT_SUPPORTED and PSA_ERROR_INVALID_ARGUMENT are acceptable outcomes (and likely a future release of mbed TLS will just use PSA_ERROR_INVALID_ARGUMENT). I think the correct behaviour for tests 224 and 225 would be to accept both error codes.

adeaarm, 2022-02-15:
Opened corresponding PR #308

jk-arm, 2022-02-16:
@adeaarm , i understand that its purely implementation-dependent, some implementations can check "SUPPORT" first and others can check "INVALID ARGUMENT", is there feedback given to Andrew about this?

adeaarm, 2022-02-16:
As you can read from the linked PR:

> The PSA Crypto API spec specifies PSA_ERROR_NOT_SUPPORTED in the case
that the supplied algorithm is not an AEAD one.

This inconsistency is an editorial error in the current version of the specification. It will be fixed in the upcoming 1.1 version, which will make it clear that as generally intended, both NOT_SUPPORTED and INVALID_ARGUMENT are compliant if the algorithm is not an AEAD algorithm.

The inconsistency between one-shot and multipart in our implementation is technically compliant, but poor implementation quality.

_Originally posted by @gilles-peskine-arm in https://github.com/ARMmbed/mbedtls/issues/5272#issuecomment-987212503_

this is going to be clarified in the 1.1 update of the spec. The assumption that the ACK tests are making (that only `PSA_ERROR_NOT_SUPPORTED` is the acceptable outcome for these cases) is too restrictive as both `PSA_ERROR_NOT_SUPPORTED` and `PSA_ERROR_INVALID_ARGUMENT` are acceptable). The comment is coming from one of the authors of the spec as feedback.

jk-arm, 2022-02-16:
thanks @adeaarm

adeaarm, 2022-02-17:
> thanks @adeaarm

Just wanted to add that in this change you can see the rework of the API documentation that is going to be published in the 1.1 release which clarifies this behaviour as well:

https://github.com/ARM-software/psa-crypto-api/pull/501/files#diff-9a65ee4a6e5b31a9bf5dd7d68753ba6f1dcc9063a06cb36b93d84095dc768df8

jk-arm, 2022-02-25:
the changes merged now, let us revisit this once the 1.1 spec investigation started

closed 2022-02-25 by jk-arm