ARM-software/psa-arch-tests #310: Expected Signature is HardCoded for Sign Hash RSA Tests

zaya-mc, 2022-03-04:
Hello,

In the test 41, we sign a hash value using RSA ( "Test psa_sign_hash - RSA KEY_PAIR PKCS1V15 SHA-256\n",
), and we try to verify it using an expected signature, but as the RSA signature will be random, it does not make sense to use a hard-coded signature, so the tests fail as expected.

If we use the signature from the sign_hash step in verify, it passes the test. 

Would you fix it or do you want us to create a PR?

Thank you.

jk-arm, 2022-03-07:
please create a PR we will review the same.

zaya-mc, 2022-03-09:
The same problem is also for test 50 (sign message) in the checkpoint 7.

jk-arm, 2022-03-29:
could you please create a PR for the same

zaya-mc, 2022-04-21:
@jk-arm sorry for delayed response. How did you get example signature, and are they tested? If we use signature created by sign function during the verify it passes, but it does not pass with the signature provided by psa tests?

We may re-use the signature from sign function in test 41, but test 42 only verify using the hard-coded signature, so we have no change to produce a new signature.

So, I am wondering whether the hard-coded signatures are tested?

zaya-mc, 2022-04-23:
Hello, it seems there is some misunderstanding in my side, I was thinking sign message, and sign hash/digest are the same thing, but it seems not. So, there is no problem with 41 and 42, 51, so they are fine. FALSE ALARM.

But; in the test 50, psa tests signs a message which should be random, and compares random signature with hard-coded value in TEST_CHECKPOINT_NUM(7). 
Herein, I am wondering do we really need this test, as it compares only expected length and signature?

athoelke, 2022-04-25:
> But; in the test 50, psa tests signs a message which should be random, and compares random signature with hard-coded value in TEST_CHECKPOINT_NUM(7).
> Herein, I am wondering do we really need this test, as it compares only expected length and signature?

Test 50 only has one non-error test scenario, and that is using `PSA_ALG_RSA_PKCS1V15_SIGN_RAW`, which is deterministic. The  randomized signature algorithms (RSA-PSS and non-deterministic ECDSA) are not currently tested in this suite.

jk-arm, 2022-04-25:
> Test 50 only has one non-error test scenario, and that is using `PSA_ALG_RSA_PKCS1V15_SIGN_RAW`, which is deterministic. The randomized signature algorithms (RSA-PSS and non-deterministic ECDSA) are [...]

@zaya-mc , hope this answers your question. thankyou @athoelke

closed 2022-06-08 by zaya-mc