ARM-software/psa-arch-tests #311: Expected Output Length for psa_cipher_finish

zaya-mc, 2022-03-14:
Hi everyone, 

We are a bit confused about the output_lenght in psa_cipher_finish().

```
 * \param[out] output_length    On success, the number of bytes
 *                              that make up the returned output.
```

Is the total bytes (update, finish etc) encrypted, or just encrypted bytes in psa_cipher_finishfunction().

There are also conflicts in tests.

In the tests;

Test 37
 - Unprocessed length : 0 (Not multiple of Block Size)
 - Total processed Length : 16
 
 But **expected output_length is zero.**
 
 
 Test 49
 - Unprocessed length : 0 (Not multiple of Block Size)
 - Total processed Length : 16
 
 But **expected output_length is 1**6 which conflicts with Test 37 expected output len.
 
 
 Any clarification here?

athoelke, 2022-03-15:
From a quick look at the code, test case 37 is checking for the length of data that is output from `psa_cipher_finish()` (see https://github.com/ARM-software/psa-arch-tests/blob/7c69c2bc6fda40ffd9ddfa7d167c3524a0ab6a78/api-tests/dev_apis/crypto/test_c037/test_c037.c#L114).

This may result in test cases that are implementation-specific, rather than implementation-independent.

The test should instead be checking that `get_update_output_length + get_finish_output_length` corresponds to the correct output size for the test case.

**Rationale:**

In the API design, the reason for `psa_cipher_finish()` step is to indicate that the input is complete and allow any remaining blocks of input to be processed (e.g. padding added or checked and removed) to return the final output data. However, the API also permits implementation flexibility and does not require that an implementation returns a specific amount of data from any individual call to `psa_cipher_update()` or `psa_cipher_finish()`. This enables implementation-specific levels of input and output buffering for the operation. The only requirement is that the concatenation of all of the returned data is identical, and corresponds exactly to the requested cryptographic algorithm applied to the inputs.

jk-arm, 2022-04-19:
@zaya-mc, in both test cases we are not really concentrating on the IV part. that's the reason you are seeing output length confusion.

athoelke, 2022-04-20:
> ... in both test cases we are not really concentrating on the IV part. that's the reason you are seeing output length confusion.

@jk-arm - however, the test cases are not implementation-agnostic (which they should be).

Currently, the test case is checking the value of `get_update_output_length`, which is not a value that is determined exactly by the specification. As I described earlier:

> The test should instead be checking that `get_update_output_length + get_finish_output_length` corresponds to the correct output size for the test case.

jk-arm, 2022-04-20:
@athoelke, yes we are making changes to your suggestion as well as the expected length in test data to account for the IV.

jk-arm, 2022-04-20:
@zaya-mc just corrected the same
https://github.com/ARM-software/psa-arch-tests/commit/0d4bf726e5d91b7418123183fe475a651df149d1

zaya-mc, 2022-04-23:
@jk-arm thank you for the fix, it solves our problem with test 37 and 49.

closed 2022-04-25 by jk-arm