ARM-software/psa-arch-tests #315: Allow using prefetched QCBOR source

gyuri-szing, 2022-04-01:
Hi,

when compiling the IAT tests, the psa-arch-test cmake scripts will clone (see [1]) the needed QCBOR dependency to CMAKE_CURRENT_BINARY_DIR. In our yocto setup network access is disabled during the yocto build phase, which makes the psa-arc-test build fail.

Would it be possible to change the psa-arch-test cmake scripts to optionally instead of cloning use a pre-fetched QCBOR dependency?

Thanks!

[1]: https://github.com/ARM-software/psa-arch-tests/blob/7c69c2bc6fda40ffd9ddfa7d167c3524a0ab6a78/api-tests/CMakeLists.txt#L523

jk-arm, 2022-04-04:
@gyuri-szing its certainly possible to have flag around CBOR cloning and i can let you point the QCBOR include path, i will come back on this

just a clarification needed, how are tfm cloning the psa-arch-test during the build when you don't have the network access?

gyuri-szing, 2022-04-04:
> just a clarification needed, how are tfm cloning the psa-arch-test during the build when you don't have the network access?

I am working on Trusted Services and TF-M is not involved. The TS build-system allows using pre-fetched source code of external components. E.g.: when running the TS build, one can pass `-DPSA_ARCH_TESTS_SOURCE_DIR=<location>` to stop cmake cloning psa-arch-test and instead use the source code from the specified location. Then one can configure yocto to clone the proper psa-arch-test version and pass the location to TS build.

jk-arm, 2022-04-05:
fixed as part of https://github.com/ARM-software/psa-arch-tests/pull/321

closed 2022-04-05 by jk-arm