ARM-software/psa-arch-tests #324: curves of size <255 are obsolete ECC algorithms and should probably be disabled.

Summer-ARM, 2022-06-13:
For the TLS protocol, [RFC 8422](https://datatracker.ietf.org/doc/html/rfc8422#section-5.1.1) deprecates many curves previously defined in [RFC 4492](https://datatracker.ietf.org/doc/html/rfc4492#section-5.1.1). That one is a mixture of non-use (curves other than secpNNNr1 weren't used for TLS) and size (curves with NNN < ~250, e.g. secp192r1 and secp224r1).

I found that ECC_SECP_R1_224 is still be used as test algorithm in psa crypto arch test. Could we use size>255 as ECC test algorithms instead of obsolete ECC algorithms?

jk-arm, 2022-07-15:
@Summer-ARM , as long as spec holds this value the psa arch test will have a testcase testing on this. if ou dont want this to be tested, please comment out the define part of the ARCH_TEST_ECC_CURVE_SECP224R1

jk-arm, 2022-11-30:
@Summer-ARM  hopefully this can be closed,

closed 2022-11-30 by jk-arm