ARM-software/psa-arch-tests #325: server test should not be enable when test policy validation.

Summer-ARM, 2022-06-14:
```
******************************************
TEST: 5 | DESCRIPTION: Testing STRICT policy with higher version
[Info] Executing tests from non-secure
[Check 1] Test STRICT policy with higher version
 psa_wait returned with invalid signal value = 0x00000010
 Checkpoint 202 : Error Code=0x00000011 
TEST RESULT: FAILED (Error Code=0x00000010)

******************************************
TEST: 6 | DESCRIPTION: Testing STRICT policy with lower version
[Info] Executing tests from non-secure
[Check 1] Test STRICT policy with lower version
 psa_wait returned with invalid signal value = 0x00000010
 Checkpoint 202 : Error Code=0x00000011 
TEST RESULT: FAILED (Error Code=0x00000010)

******************************************
TEST: 7 | DESCRIPTION: Testing RELAX policy with higher version
[Info] Executing tests from non-secure
[Check 1] Test RELAX policy with higher version
 psa_wait returned with invalid signal value = 0x00000010
 Checkpoint 202 : Error Code=0x00000011 
TEST RESULT: FAILED (Error Code=0x00000010)

******************************************
TEST: 10 | DESCRIPTION: Testing un-specified version_policy with higher version
[Info] Executing tests from non-secure
[Check 1] Test un-specified version_policy with higher version
 psa_wait returned with invalid signal value = 0x00000010
 Checkpoint 202 : Error Code=0x00000011 
TEST RESULT: FAILED (Error Code=0x00000010)

******************************************
TEST: 11 | DESCRIPTION: Testing un-specified version_policy with lower version
[Info] Executing tests from non-secure
[Check 1] Test un-specified version_policy with higher version
 psa_wait returned with invalid signal value = 0x00000010
 Checkpoint 202 : Error Code=0x00000011 
TEST RESULT: FAILED (Error Code=0x00000010)
******************************************
```

All these related policy test case should not enable server test, otherwise server partition received signal will be wrong.
Take 05 test for example:
in test_i005_client_tests, it passed wrong service version (SERVER_STRICT_VERSION_SID, SERVER_STRICT_VERSION_VERSION + 1), tfm spm will immediately check out this error and directly returned the error code to caller instead of enter requested partition service function. While the server test is still waiting for SERVER_STRICT_VERSION_SIGNAL (it won't be got since tfm return error code to caller to refuse the connection). This will cause server test failed and eventually cause the 05 test case failed.
In my idea, the test purpose is only to test policy validation, so no need to enable serve test which is always be failure since won't get the desired signal.

Suggest changes 
`status = val->execute_non_secure_tests(TEST_NUM, test_i005_client_tests_list, TRUE); `
should changed to:
 `status = val->execute_non_secure_tests(TEST_NUM, test_i005_client_tests_list, FALSE);`

jk-arm, 2022-06-21:
@Summer-ARM this was the failing test cases from the beginning, having said that we will make a investigation soon on this.

jk-arm, 2022-06-21:
approved the suggestion and corrected in main branch

closed 2022-06-21 by jk-arm