ARM-software/psa-arch-tests #326: FF test 27 is not design correctly.

Summer-ARM, 2022-06-14:
In this test case, the purpose is to test all subsequent calls to psa_call() with the same connection handle will will immediately return PSA_ERROR_PROGRAMMER_ERROR when it returns PSA_ERROR_PROGRAMMER_ERROR for the first time.

In the first time, it calls psa_call() [here](https://github.com/ARM-software/psa-arch-tests/blob/main/api-tests/ff/ipc/test_i027/test_i027.c#L81),  it will return PSA_ERROR_PROGRAMMER_ERROR. It is not becuase any error is happened, it is just because service return this value [here](https://github.com/ARM-software/psa-arch-tests/blob/main/api-tests/ff/ipc/test_i027/test_supp_i027.c#L56).
The second time, it wants to call psa_call() agian [here](https://github.com/ARM-software/psa-arch-tests/blob/main/api-tests/ff/ipc/test_i027/test_i027.c#L107). This time it is getting to the [service disconnect operation ](https://github.com/ARM-software/psa-arch-tests/blob/main/api-tests/ff/ipc/test_i027/test_supp_i027.c#L65)(which is blocked point at last time). So, it will get wrong message type (PSA_IPC_CALL instead of PSA_IPC_DISCONNECT) which cause the test failed.

The log is:
```
TEST: 27 | DESCRIPTION: Testing connection termination at PSA_IPC_CALL
[Info] Executing tests from non-secure
[Check 1] Test connection termination at PSA_IPC_CALL
 psa_get failed for disconnect massage
 Checkpoint 203 : Error Code=0x00000011 
 Disconnect request failed
 Call should have returned PSA_ERROR_PROGRAMMER_ERROR. Status = 0x00000000
In server_partition, Control shouldn't have reach here
 
TEST RESULT: SIM ERROR (Error Code=0x00000011)
```

In my opinion, this test is not design correct, could you help to refine it?

jk-arm, 2022-06-23:
the latest code with the latest TFM is passing this test. so i will wait for TFM to confirm the same.

jk-arm, 2022-09-01:
waiting for longer time, hence closing this ticket and please raise again if you see this again

closed 2022-09-01 by jk-arm