ARM-software/psa-arch-tests #333: Test cases to probe integer overflow handling

athoelke, 2022-11-10:
Some APIs have integer inputs with defined pre-conditions on valid values. Naïve implementation of condition checks in an implementation may trigger integer overflow resulting in undefined and possibly incorrect behavior of the API.

For example, `psa_ps_set_extended()` has a `data_offset` and `data_length` inputs, and the stored item has an existing `capacity`: these values must satisfy `data_offset + data_length <= capacity`. As the left hand side values are both caller controlled, this sum cannot be evaluated without considering the possibility of integer overflow. It would be good to have test cases that supplied parameter values that would overflow in implementations that did not take this into account.

Similar concerns exist for APIs where the caller provides a size of an item that must be allocated or created. If the implementation adds to, or multiplies the caller value without check for overflow, then it can result in incorrect operation, and a possible vulnerability. It would be valuable to test such APIs with extreme values (e.g. MAX_UINT32).

jk-arm, 2022-11-11:
@athoelke, i believe the kind of testing asked here falls under the DV or UT test cases, the psa-arch-test is here to test the invariant behavior of the specification, if the specification talks about the overflow and it's error code check then we can bring that under psa-arch-test.

jk-arm, 2022-11-11:
@athoelke the spec does not talk about the overflow of the below condition
> data_offset + data_length <= capacity, so thats the reason we are only checking the normal INVALID_ARGUMENTS where the data_len + offset goes greater than the capacity. fyi, this we could not exercise in the TFM because the psa_ps_set_extended() is not implemented.

athoelke, 2022-11-11:
The spec provides the constraint on the input parameters, with a required error condition response. The spec is describing the constraint in _mathematical_ terms, where 'overflow' does not exist. So there is no reason for the specification to specifically describe the condition for an overflow, when that is already covered by the constraint already provided.

If the specification had a SRA (i.e. like the Crypto API), then there is a threat related to logic errors in the implementation when handling attacker-controlled (caller provided) values. The SRA would delegate mitigation of such threats to the implementation.

So this could be considered as 'not a functional correctness test'. However, if a call of `psa_ps_set_extended(uid, 1, MAX_UINT32, buf)` did not report `PSA_ERROR_INVALID_ARGUMENT` then it is failing to meet the spec, which requires that this is an error because `1 + MAX_UINT32 == 0x100000000 > capacity`.

Similarly, the spec requires that if a caller is asking for more resource than is available, it returns an appropriate error, such as `PSA_ERROR_INSUFFICIENT_XXX`. For example, a test case such as `psa_ps_create(uid, MAX_UINT32, 0)` should fail with an error, rather than succeed because of a logic/integer error within the implementation.

jk-arm, 2022-11-11:
@athoelke agreed, psa arch test team will add the  `So this could be considered as 'not a functional correctness test'. However, if a call of psa_ps_set_extended(uid, 1, MAX_UINT32, buf) did not report PSA_ERROR_INVALID_ARGUMENT then it is failing to meet the spec, which requires that this is an error because 1 + MAX_UINT32 == 0x100000000 > capacity.` sometime in jan as we are busy on the other project release activity

athoelke, 2022-11-11:
There is one test case within one test program for storage that tries this, IDX12 in s008 (https://github.com/ARM-software/psa-arch-tests/blob/main/api-tests/dev_apis/storage/test_s008/test_s008.c#L72) uses MAX_UINT32 as the offset in `psa_xxx_get()`, with a small length.

But the inverse (valid offset with MAX_UINT32 length) is not attempted. This might result in imp-def behavior because the buffer being passed is not all accessible to the caller -  so the framework might respond with a PROGRAMMER_ERROR, or the client might be terminated, or might have its memory overwritten. However, if the call reaches the service, the service should handle the scenario gracefully and only return exactly the data that is available within the item. In no situation should the storage or storage service be corrupted, or the service report more or less than the data that is stored in the item.

athoelke, 2022-11-11:
> sometime in jan as we are busy on the other project release activity

That is good, thank you. This has been a long-standing item in the Secure Storage API issues, and I am reviewing those now for migration to the psa-api repository. This one was relevant for the test suite, rather than the API.

avinaw01-arm, 2024-06-11:
Hi @athoelke,

We have made resolution changes for this issue and included them in a Pull Request to this repository. You can find the pull request link here: https://github.com/ARM-software/psa-arch-tests/pull/379.
I request you to please review these changes from your end as well, and let us know if they resolve the issue satisfactorily.

Thank You,
Avi.

athoelke, 2024-06-12:
This is a case where more than one error code applies to the parameters in the call:

* The offset and size of the request are not valid for the data item.
* The client memory buffer for the request is invalid.

Depending on the implementation, these may be evaluated in a different order, and the latter may only be detected when the program faults, rather than via any explicit memory buffer validation logic.

With TF-M, the client call to the storage API is routed to the storage service by the SPM, using the PSA FF-M API [FF-M]. (This framework is defined in *Arm® Platform Security Architecture Firmware Framework* https://developer.arm.com/documentation/den0063).

In a call to `psa_ps_get()` (documented [here](https://arm-software.github.io/psa-api/storage/1.0/api/api.html#c.psa_ps_get)), the pair of values {`p_data`, `data_size`} is considered to be a reference to a writable buffer in the caller. This is because the call is requesting that much data, and so the implementation expects that the caller has a buffer of that size available.

When that output buffer reference is transferred to the service via the `psa_call()` (see p69 in FF-M), the SPM is required to validate each memory buffer that the client provides (i.e. the entire buffer must be entirely accessible by the caller). An invalid memory buffer is a Programmer Error, and this can either result in the call not returning at all, the termination of the caller, or a `PSA_ERROR_PROGRAMMER_ERROR` return value. Further details of the requirements for client memory buffers is in FF-M §3.3.5 (p41).

TF-M implements that logic when processing calls from clients, in `spm_associate_call_params()` on [lines 134-149 of psa_call_api.c](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/heads/main/secure_fw/spm/core/psa_call_api.c#134).

The call never reaches the storage service, so any risk of integer overflow in the service is not tested in this case.

athoelke, 2024-06-12:
I see that I partially discussed the scenario of `data_size == MAX_UINT32` in my earlier comment (https://github.com/ARM-software/psa-arch-tests/issues/333#issuecomment-1311508053).

avinaw01-arm, 2024-07-23:
Hi @athoelke,
We have pushed in a new commit in the resolution P.R.: https://github.com/ARM-software/psa-arch-tests/pull/379.
Tried incorporating our meeting discussions in the same. Please review and let us know for any comments.

Thanks & Regards,
Avi.

avinaw01-arm, 2024-07-25:
Merged P.R. https://github.com/ARM-software/psa-arch-tests/pull/379, as a resolution of this Issue.
Please have a look, if required.

Thank you,
Avi.

closed 2024-07-25 by avinaw01-arm