ARM-software/psa-arch-tests #336: Only /n line endings instead of /r/n

mswarowsky, 2022-12-05:
Hei,

While doing release testing of the TF-M 1.7-rc2 on nRF5340 and nRF9160 I noticed that the line endings sent by the tests just have a `\n` not a `\r\n` causing format issues displaying the output on several serial terminals in linux, with default settings. 

The test I have seen this behavior was Crypto/Storage/Initial Attestation.

So might be a good choice to replace all `\n` to `\r\n` in the logging or uart driver. As I have only nRF platforms available I don't know if that is already been done for other platforms.

jk-arm, 2022-12-15:
@mswarowsky currently the printing functionality is left to platform interface implementation, meaning the partner platform abstraction layer implementation will take care. please refer to the "pal_print"

jk-arm, 2022-12-20:
@mswarowsky please let me know, so that i can move this ticket to appropriate state or clsed.

mswarowsky, 2022-12-20:
Yeah closed it

closed 2022-12-23 by jk-arm