ARM-software/psa-arch-tests #339: Test failure c021

balint-dobszay-arm, 2023-01-12:
Updating from Mbed TLS 3.1.0 to 3.2.1 in [Trusted Services](https://git.trustedfirmware.org/TS/trusted-services.git/) causes a new test failure:
```
TEST: 221 | DESCRIPTION: Testing crypto key derivation APIs | UT: psa_key_derivation_output_key
[Info] Executing tests from non-secure
[Check 1] Test psa_key_derivation_output_key - Key
[Check 2] Test psa_key_derivation_output_key - Info
[Check 3] Test psa_key_derivation_output_key - Salt
[Check 4] Test psa_key_derivation_output_key - Greater Capacity than available
[Check 5] Test psa_key_derivation_output_key - ECC Public key
[Check 6] Test psa_key_derivation_output_key -  ECC keypair (wrong key length)
	Failed at Checkpoint: 14
	Actual: 0
	Expected: -135

TEST RESULT: FAILED (Error Code=0x1)
```

Earlier there was a PR in this repo (https://github.com/ARM-software/psa-arch-tests/pull/312) to change the expected result for this test case from `PSA_ERROR_INVALID_ARGUMENT` to `PSA_SUCCESS`: https://github.com/ARM-software/psa-arch-tests/pull/312/commits/e6bac082c44c982e750d61cf6fb6d4e3658ccd71#diff-dcb8deacc729e1ff043f05e55ee8ad6ac6532157eb214400f4d28b102e46ab6bR113

However, this PR was closed without merging, so the mentioned change didn't get to the main branch: https://github.com/ARM-software/psa-arch-tests/blob/main/api-tests/dev_apis/crypto/test_c021/test_data.h#L113

Could you please double check if the result expected by the arch test is correct for this case?

balint-dobszay-arm, 2023-01-20:
FYI @bensze01

jk-arm, 2023-01-23:
@balint-dobszay-arm , as i mentioned offline we dont really clone the mbedtls separately, we will clone the TFM and expect tfm to clone the compatible mbed-crypto & mbedtls package,  fyi, with respect to the latest TFM available testcase 21 is passing.

paul-elliott-arm, 2023-01-25:
@jk-arm I don't see how this failing test is an accurate test. It claims to be attempting to derive a key into an invalid length keypair (or at least that is how I interpret the test description), however the bitlength of the `PSA_KEY_TYPE_ECC_KEY_PAIR(PSA_ECC_FAMILY_SECP_R1)` provided is 224, which **is** supported, both under the PSA Spec and by MbedTLS (see: https://arm-software.github.io/psa-api/crypto/1.1/api/keys/types.html?highlight=psa_ecc_family_secp_r1#c.PSA_ECC_FAMILY_SECP_R1)

This test used to pass because we had not actually implemented ECC key derivation, but now that we have this test passes, because it is an invalid negative test as far as I can see.

There are two potential solutions here - either remove this test completely, or correct it to an invalid bit length. Which would you prefer?

balint-dobszay-arm, 2023-02-02:
@jk-arm any updates on this?

jk-arm, 2023-02-13:
@balint-dobszay-arm sorry for the late reply, we got stuck on other priority work, we will look into this week.

jk-arm, 2023-02-13:
@balint-dobszay-arm , i think its better i will change the expected bit lenght from 224 to 220, as 224 is valid.
thankyou

jk-arm, 2023-02-14:
@balint-dobszay-arm  done the changes and merged in main

balint-dobszay-arm, 2023-02-15:
Thank you, this solved the issue.

closed 2023-02-15 by balint-dobszay-arm