ARM-software/psa-arch-tests #344: Compilation error when using cmake 3.25 and Ninja.

joerchan, 2023-03-04:
How to reproduce:

```
# make sure cmake 3.25 is installed
# pip3 install --user cmake==3.25.2
cmake --version
cmake version 3.25.2

# Configure cmake
cd <trusted-firmware-m>
cmake -S . -B b -GNinja -DTFM_PLATFORM=nordic_nrf/nrf9160dk_nrf9160 -DTEST_PSA_API=CRYPTO 

# Build
ninja -C b

# Error output
ninja: Entering directory `b'
[167/492] Performing configure step for 'psa_generate_database'
-- Configuring done
-- Generating done
-- Build files have been written to: trusted-firmware-m/b/tf-m-tests/app/psa_api_tests/src/psa_generate_database-build
[172/492] Performing install step for 'psa_generate_database'
FAILED: tf-m-tests/app/psa_api_tests/src/psa_generate_database-stamp/psa_generate_database-install 
trusted-firmware-m/b/tf-m-tests/app/psa_api_tests/src/psa_generate_database-stamp/psa_generate_database-install 
cd trusted-firmware-m/b/tf-m-tests/app/psa_api_tests/src/psa_generate_database-build && /usr/bin/cmake --install . && \
/usr/bin/cmake -E touch trusted-firmware-m/b/tf-m-tests/app/psa_api_tests/src/psa_generate_database-stamp/psa_generate_database-install
-- Install configuration: ""
CMake Error at cmake_install.cmake:54 (file):
  file INSTALL cannot find
  "trusted-firmware-m/b/tf-m-tests/app/psa_api_tests/src/psa_generate_database-build/target_database.h":
  No such file or directory.

```

This affects the Zephyr Project and the nRF Connect SDK, which all use cmake and ninja.
This version of Cmake will be installed using when using 'apt' on Ubuntu.

jk-arm, 2023-03-08:
Hi could you please try the cmake with unix makefile generation?, looks like ninja build issue on 3.25

joerchan, 2023-03-10:
@jk-arm I just tried, Makefiles, and on first try this works.

However this could be simply because there is a race-condition that is not discovered when things are ordered and slow.
I therefore tried again with `make -j10` and I got this:

```
trusted-firmware-m/b/lib/ext/psa_arch_tests/api-tests/val/nspe/val_interfaces.h:121:10: fatal error:
 test_entry_fn_declare_list.inc: No such file or directory
  121 | "include @test_entroy_fn_declare_list.inc"
```

I tried to build with `ninja -C b -j1` and I still get the same error, so this one is more consistent.
I tried Makefiles on cmake 3.24.3, and this works.

It looks to me that psa-arch-tests are not following correct generated file-level dependency in cmake, and that this can manifest in different errors based on the build system used when using cmake 3.25.

jk-arm, 2023-03-14:
@joerchan, we could clearly see that between 3.24 and 3.25 nothing has changed from the proj cmake file point of view, we had a similar issue reported internally and later reporter found something to do with the 3.25 ninja.

joerchan, 2023-03-14:
Yeah, cmake -G"Unix Makefiles" has a different error, if you compile with more than one job.
But it is on the same type of issue. Compilatior error because of generated files not being written. I.e the dependency tree has issues.

joerchan, 2023-03-23:
It's the same as above, but without the '-GNinja' argument. This takes the default on my system which is 'Unix Makefiles".

```
# make sure cmake 3.25 is installed
# pip3 install --user cmake==3.25.2
cmake --version
cmake version 3.25.2

# Configure cmake
cd <trusted-firmware-m>
# Optionally: cmake -S . -B b -G"Unix Makefiles" -DTFM_PLATFORM=nordic_nrf/nrf9160dk_nrf9160 -DTEST_PSA_API=CRYPTO 
cmake -S . -B b -DTFM_PLATFORM=nordic_nrf/nrf9160dk_nrf9160 -DTEST_PSA_API=CRYPTO 

# Build
make -C b -j10
```

jk-arm, 2023-04-20:
@joerchan sorry for the gap, we had other proj deliveries, fyi, i have tried with cmake 3.25.3 and it worked right

joerchan, 2023-04-26:
@jk-arm Thank you for getting back to me. I'm not able to try 3.25.3, since pip only has 3.25.0 and 3.25.2.
It strange that they don't have .1 and .3.

I've tested with 3.26.0 and there it does not appear, so looks like the issue has been fixed.

I think I will just write this in our known issues section.

Your suggested changes did not fix the issue I'm afraid.

cneveux, 2023-04-26:
@jk-arm, @joerchan,

I'm facing the same issue. It looks like the following step is just doing the cmake --install and it's not doing the build.
Removing the `BUILD_COMMAND ""` solve the problem.

```
ExternalProject_Add(
        ${PSA_TARGET_GENERATE_DATABASE}
        PREFIX ${CMAKE_CURRENT_BINARY_DIR}
        DOWNLOAD_COMMAND ""
        UPDATE_COMMAND ""
        PATCH_COMMAND ""
        BUILD_COMMAND ""
        SOURCE_DIR "${PSA_ROOT_DIR}/tools/scripts/target_cfg"
	CMAKE_ARGS -DPYTHON_EXECUTABLE=${PYTHON_EXECUTABLE}
		-DOUT_DIR=${CMAKE_CURRENT_BINARY_DIR}
		-DTARGET=${TARGET}
		-DGENERATOR_FILE=${PSA_TARGET_CONFIG_HEADER_GENERATOR}
		-DINCLUDE_DIR=${PSA_ROOT_DIR}/val/common
		-DTARGET_CONFIGURATION_FILE=${TARGET_CONFIGURATION_FILE}
		-DTGT_CONFIG_SOURCE_C=${TGT_CONFIG_SOURCE_C}
		-DOUTPUT_HEADER=${OUTPUT_HEADER}
		-DDATABASE_TABLE_NAME=${DATABASE_TABLE_NAME}
		-DDATABASE_TABLE_SECTION_NAME=${DATABASE_TABLE_SECTION_NAME}
		-DTARGET_HEADER_GEN_INCLUDE_PATHS=${TARGET_HEADER_GEN_INCLUDE_PATHS}
	LIST_SEPARATOR |
        TEST_COMMAND ""
)
```

joerchan, 2023-04-26:
Thanks @cneveux , I can confirm that that fixes the issue for me as well.

joerchan, 2023-04-26:
@jk-arm I think not having BUILD_ALWAYS TRUE is also an issue that should be fixed.

jk-arm, 2023-04-27:
@cneveux thankyou for the information
@joerchan , i have updated both the information as below, the same i will push into the mainline
ExternalProject_Add(
        ${PSA_TARGET_GENERATE_DATABASE}
        PREFIX ${CMAKE_CURRENT_BINARY_DIR}
        DOWNLOAD_COMMAND ""
        UPDATE_COMMAND ""
        PATCH_COMMAND ""
        BUILD_ALWAYS TRUE
        SOURCE_DIR "${PSA_ROOT_DIR}/tools/scripts/target_cfg"
        CMAKE_ARGS -DPYTHON_EXECUTABLE=${PYTHON_EXECUTABLE}
                -DOUT_DIR=${CMAKE_CURRENT_BINARY_DIR}
                -DTARGET=${TARGET}
                -DGENERATOR_FILE=${PSA_TARGET_CONFIG_HEADER_GENERATOR}
                -DINCLUDE_DIR=${PSA_ROOT_DIR}/val/common
                -DTARGET_CONFIGURATION_FILE=${TARGET_CONFIGURATION_FILE}
                -DTGT_CONFIG_SOURCE_C=${TGT_CONFIG_SOURCE_C}
                -DOUTPUT_HEADER=${OUTPUT_HEADER}
                -DDATABASE_TABLE_NAME=${DATABASE_TABLE_NAME}
                -DDATABASE_TABLE_SECTION_NAME=${DATABASE_TABLE_SECTION_NAME}
                -DTARGET_HEADER_GEN_INCLUDE_PATHS=${TARGET_HEADER_GEN_INCLUDE_PATHS}
        LIST_SEPARATOR |
        TEST_COMMAND ""
)

closed 2023-04-28 by jk-arm