ARM-software/psa-arch-tests #345: Improvement: allow disabling tests

gyuri-szing, 2023-03-14:
Please consider implementing a solution which allows disabling a list of test cases. Ideally this should be run-time configuration, but a compile-time configuration would fulfill our requirements too.  
This would allow
* configuring the test to match the capabilities of the implementation
* and to temporarily disable tests for the time period when the crypto implementation and psa-acs is incompatible (this can happen when the crypto implementation is updated.)

**Background**
Trusted Services will never implement hash_suspend/resume crypto operations and currently we have to patch psa-acs  to disable these test cases. Although *-DSUITE_TEST_RANGE* would work in this case since the tests cases are at the end of the range (62, 63), this solution is not future proof.

Adopting a test framework like [CppUTest](http://cpputest.github.io/) would add test case selection with command line arguments based on string matching test name, or test group name. In addition, it would add standard output format (xunit, teamcity), and  native implementation for gathering test cases. This would allow simplifying the build process and eliminating gen_tests_list.py

Of course there are other less expensive options to implement this.

gyuri-szing, 2023-04-03:
@jk-arm 
> you can make any test case skip, which will not get executed as part of the regression. this is the compile-time solution,

Can you please tell me what parameter I need to pass to cmake to skip a specific test?

IMHO: this configuration is "static" meaning I need to modify the file on disk to disable a test case. What I am looking for is a "dynamic" configuration where I don not need to modify a file to change test configuration. In comparison _-DSUITE_TEST_RANGE_ is dynamic in this sense. 

What's wrong with a configuration method where files on disk need to be modified?
- For example if I check the status of the repository in my CI, the repository will be "dirty". Only manual investigation will be able to tell what changes have been made and if the psa-acs version used is valid. psa-acs is used for certification and using a modified version might be problematic.
- Another challenge is ownership of psa-acs configuration vs. ownership of setting the configuration. The configuration is owned by the project which uses psa-acs for testing the implementation (e.g. tf-m, Trusted Services). In case of some integration systems, ownership of retrieving and patching components (ownership of setting the configuration) is at the integration system. (The aim is to have all inputs defined at a shingle place for simplicity, efficiency and security.) This usually means means psa-acs configuration (patch files) need to be duplicated in the integration system. Ownership of build options in turn can stay at the project owning the psa-acs configuration, and thus build options are a better method for configuration.
- Runtime configuration adds even more flexibility. It allows using a single executable built for a specific target environment to test multiple configurations of the implementation.

jk-arm, 2023-04-03:
@gyuri-szing, i get your point of view, i can take this as a next release request which is planned in may'23, introducing the build macro involves some development and testing. we will try to provide before that based on the resource availability.

or else if you needed it urgently then you can create a pull request where we can review and merge it.

desmetbruno, 2023-04-03:
@jk-arm @gyuri-szing 
I have created PR https://github.com/ARM-software/psa-arch-tests/pull/347 that implements a solution for the issue described above. Please feel free to share your comments.

gyuri-szing, 2023-04-03:
> @jk-arm @gyuri-szing I have created PR #347 that implements a solution for the issue described above. Please feel free to share your comments.

Thanks, will take a look next week.

gyuri-szing, 2023-04-17:
@jk-arm From our side Gábor tested the modification and things look good, and after #347  is merged, this issue can be closed. Thanks!

jk-arm, 2023-04-17:
Hi ,

I am Just waiting for the architect suggestion related to licensing and some of the other quries.
I will come back ASAP

Regards,
JK

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

jk-arm, 2023-04-26:
@gyuri-szing i ahve made some suggestion to the pull request, could you please look at that?

gyuri-szing, 2023-04-26:
> i ahve made some suggestion to the pull request, could you please look at that?

@jk-arm I can only see 2-3 week old comments at #347. Unless I am looking at the wrong place , from our side the change is ok. 
(FYI: @gabor-toth-arm is sitting in my team and he was tasked to create a POC on our side ([please find it here if interested](https://review.trustedfirmware.org/c/TS/trusted-services/+/20532)).  )

jk-arm, 2023-05-02:
@gyuri-szing i was talking about the comment which mentioned in the pull request https://github.com/ARM-software/psa-arch-tests/pull/347

@desmetbruno can we please not have the two pal apis, can we have it in a single API?, 
i was more trying to do it as below with singlw api.
uint32_t crypto_custom_list[] = {1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,\
                                 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,\
                                 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1};
uint32_t initial_attestation_custom_list[] = {1};
uint32_t storage_custom_list[] = {1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1};

/**
 *   @brief    - Tells if a test is enabled on platform
 *   @param    - test_id : Test ID
 *   @return   - TRUE/FALSE
**/
bool_t pal_is_test_enabled(test_id_t test_id)
{
#ifdef CRYPTO
    uint32_t crypto_list_size = sizeof(crypto_custom_list)/sizeof(crypto_custom_list[0]);
    if(test_id > crypto_list_size)
        return 1;
    else
        return crypto_custom_list[(test_id%MAX_TEST_NUMBER)-1]

#elif STORAGE
    uint32_t storage_list_size = sizeof(storage_custom_list)/sizeof(storage_custom_list[0]);
    if(test_id > storage_list_size)
        return 1;
    else
        return storage_custom_list[(test_id%MAX_TEST_NUMBER)-1]
#elif INITIAL_ATTESTATION
    uint32_t attestation_list_size = sizeof(initial_attestation_custom_list)/sizeof(initial_attestation_custom_list[0]);
    if(test_id > attestation_list_size)
        return 1;
    else
        return initial_attestation_custom_list[(test_id%MAX_TEST_NUMBER)-1]
#else
    return 1;
#endif

}

desmetbruno, 2023-05-03:
@jk-arm 
- pal_set_custom_test_list() is called by the main() of the test application to set the custom test list. See example below.
- pal_is_test_enabled() is called by VAL to check if the test should be run or not

With your proposal, how do you update crypto_custom_list[], initial_attestation_custom_list[] or sotrage_custom_list[] at runtime?
Customizing the test list at runtime is one of the goals of PR#347.

I find it more convenient to reuse test_entry_list.inc which gives the exact test ID built by gen_tests_list.py. It has also the advantage of making the function 'suite' agnostic.
What do you think?

Example of main.c

size_t val_get_test_list(uint32_t *test_id_list, size_t size);
void pal_set_custom_test_list(char *custom_test_list);

int main(int argc, char **argv)
{
	uint32_t ret;
	char *custom_test_list = NULL;

[...]

	/*
	 * Set the variable custom_test_list here.
	 * In case of Linux, it may be the content of a text file.
	 * For any platform, it can be set based on platform capabilities
	 * (for example a hard-coded constant variable in this file).
	 */

	pal_set_custom_test_list(custom_test_list);

	ret = val_entry();

[...]

	if (custom_test_list)
		free(custom_test_list);

	return ret;
}

jk-arm, 2023-05-03:
HI,

I was just writing the logic only to propose a single PAL API, yes the array will get updated from the text file incase of operating system which supports the file system.

my main idea is to have a single PAL api, not two.

Regards,
JK

jk-arm, 2023-05-05:
closing this ticket

closed 2023-05-05 by jk-arm