ARM-software/psa-arch-tests #346: Make hardcoded attestation key optional

balint-dobszay-arm, 2023-03-17:
In the Linux platform configuration (`api-tests/platform/targets/tgt_dev_apis_linux/nspe/pal_config.h`) the `PLATFORM_OVERRIDE_ATTEST_PK` option is enabled by default. This is problematic since not all platforms use the hardcoded attestation key, and there is no way of disabling this except patching the psa-arch-tests repo which is not ideal.

Please provide a way to make this feature optional. E.g.:
```
#ifndef PLATFORM_HAS_ATTEST_PK
#define PLATFORM_OVERRIDE_ATTEST_PK
#endif
```
This way the default behaviour is unchanged, but platforms using a custom attestation key could work without modifying the psa-arch-tests source.

jk-arm, 2023-03-31:
@balint-dobszay-arm its done, the suggestion is incorporated

closed 2023-03-31 by jk-arm

balint-dobszay-arm, 2023-06-26:
Hi, yes that makes sense, the declaration should be part of some header file in this repo. Please open a new issue or PR for this.