ARM-software/psa-arch-tests #376: Initial Attestation test private key

piotrParf, 2024-05-17:
For initial attestation test in every target there is provided always the same hardcoded public key. Where i\I can find the private key for it? Is it something well known?

avinaw01-arm, 2024-05-18:
I believe you're talking about `psa_algorithm_t key_alg = PSA_ALG_ECDSA(PSA_ALG_SHA_256);` public key. Hope I'm getting it right?

piotrParf, 2024-05-29:
I am just asking about the key from targets for example in file:
/psa-arch-tests/api-tests/platform/targets/tgt_dev_apis_tfm_nrf5340/nspe/pal_attestation_config.h

> `static const struct ecc_public_key_t attest_public_key = {
     /* Constant byte */
     0x04,
     /* X-coordinate */
     {0x79, 0xEB, 0xA9, 0x0E, 0x8B, 0xF4, 0x50, 0xA6,
      0x75, 0x15, 0x76, 0xAD, 0x45, 0x99, 0xB0, 0x7A,
      0xDF, 0x93, 0x8D, 0xA3, 0xBB, 0x0B, 0xD1, 0x7D,
      0x00, 0x36, 0xED, 0x49, 0xA2, 0xD0, 0xFC, 0x3F,
     /* Y-coordinate */
      0xBF, 0xCD, 0xFA, 0x89, 0x56, 0xB5, 0x68, 0xBF,
      0xDB, 0x86, 0x73, 0xE6, 0x48, 0xD8, 0xB5, 0x8D,
      0x92, 0x99, 0x55, 0xB1, 0x4A, 0x26, 0xC3, 0x08,
      0x0F, 0x34, 0x11, 0x7D, 0x97, 0x1D, 0x68, 0x64},
};
`

we have a public key used to check the attestation token. The signature of the token should be generated with a private key from which this public key is derived. Do we know that? Or I am missing something?

avinaw01-arm, 2024-06-17:
Hi @piotrParf,

Generally, the private key is held by the implementation side of the security architecture. For our internal development & testing, we use the '**_an521_**' and '**_Musca_b1_**' platforms, with **Trusted Firmware-M (TF-M)** as the implementation. Therefore, the private key is held by this reference platform and the corresponding public key, derived from this private key, is already hard-coded in the Configuration files. If you want to view the private key-part for this pair, you can check somewhere in the TF-M repo. here: https://www.trustedfirmware.org/projects/tf-m/. According to my understanding, it is not possible to derive a private key directly from a public key. There are specific procedures and requirements for obtaining the private key, but it cannot be determined solely from knowing public key.

Also, our code provides an option to avoid using the hard-coded public key. You can disable the macro `PLATFORM_OVERRIDE_ATTEST_PK` in the file: `<syscomp_psa_m/api-tests/platform/targets/<target_name>/nspe/pal_config.h` to derive the public key directly from the implementation platform during compilation time. For example, if you see this file for targets '**_nrf5340_**' and '**_nrf9160_**', the `PLATFORM_OVERRIDE_ATTEST_PK` macro has been disabled to derive the public key directly at compilation time.

avinaw01-arm, 2024-06-17:
I hope this explanation resolves your query. If not, please let us know if you have any further doubts. Otherwise, we can close this issue.

avinaw01-arm, 2024-06-25:
@piotrParf, Any query/update on this? I hope the explanation provided is satisfactory.
Please let us know so that we can proceed with closing this issue.

Regards,
Avi.

piotrParf, 2024-06-26:
For me it is clear. Thanks. I will look for the private key in TFM repo. If not I will switch to derived ones.

avinaw01-arm, 2024-06-26:
Alright, good to know! We'll close this issue now.

closed 2024-07-08 by avinaw01-arm