ARM-software/psa-arch-tests #382: psa-arch-tests and mbedtls implementation

zaya-mc, 2024-08-17:
Hello, 

It is not an issue, but is there any recommendation for the tested mbedtls versions with psa arch tests?

All we need is a PSA API demonstration and psa implementation in mbedtls software implementation may differ.

We tried mbedTLS v3.6.0 LTS, but could not success yet, so digging out the problems.
Also, in the development branch, it seems mbedtls separates psa implementation under tf-psa-crypto.

We could dig out the problems but there would be a map between psa-arch-test, tested mbedtls versions etc as psa_arc_tests needs some headers from mbedtls?

These version differences can be really time-consuming.

Thank you in advance.

jk-arm, 2024-08-18:
Please try with mbedtls 3.5.2, which is the tested version from psa-arch-test

Regards,
JK

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

avinaw01-arm, 2024-09-17:
Hi @zaya-mc,
Have you tried the implementation with MbedTLS v3.5.2 version? I hope this resolves the issue.
Please let us know if you need further assistance, or if we can now close this issue.

Thanks & Regards,
Avi.

avinaw01-arm, 2024-09-25:
Hi @zaya-mc,
I hope the resolution provided above addresses your issue. If you have any further questions, please let us know; otherwise, we will proceed to close this issue.

avinaw01-arm, 2024-09-30:
@zaya-mc, we are closing this issue. Please reach out to us if you have any further queries on the same.

Thanks & Regards,
Avi.

closed 2024-09-30 by avinaw01-arm

zaya-mc, 2024-12-06:
@avinaw01-arm @jk-arm  sorry guys, we could not chase this ticket. 

Now, we are trying v3.5.2, but it seems it does not have "PSA Attestation API"  headers under the include/PSA/ directory. Also, there is no reference to PSA Initial Attestation Implementation. 

We found a reference Initial Attestation API and implementation under TF-m, but we are not using psa API with TF-m.
We might copy this implementation, but it is coupled with headers like "tfm_*.h"

There  are PSA Attestation API headers in the psa-arch-test repo as well but conflicts with mbedtls PSA headers :) 

So, while PSA has been a standard since 2019, we would expect some mature support.
We achieved PSA Certification 2019, and it is easier than now.

So, guys, to summarise and pass all crypto, attestation and SS psa-arch-test
1. what is the mature psa-arch-test version
2. What mbedtls version would you recommend for the implementation of PSA Crypto and Attestation API headers and implementations?
3. If attestation headers and API implementation are no longer available in mbedtls, is there any attestation API/Implementation recommendation? Is TF-m the only potential API and implementation?

avinaw01-arm, 2024-12-17:
Hi @zaya-mc,

Thank you for reaching out to us. Below are some points addressing your queries:

'**psa-arch-tests**' framework do not validate direct support for implementing 'Initial Attestation' suite on Mbed-TLS libraries. As an alternative, you can refer to Trusted Firmware-M (TF-M) implementation, which is publicly available and serves as a reference implementation for running the suite. Answering to your queries in order:

1. The latest Release **v1.7** is a mature version where all suites are expected to run successfully with TF-M/Mbed-TLS, with the exception of PAKE tests; as these tests have been validated with Oberon libraries.
2. As mentioned earlier, the stable version of Mbed-TLS validated from our side is v3.5.2.
3. To our knowledge, TF-M is the only platform we have used for validating the 'Initial Attestation' suite. For specific Crypto-suite, you can utilize TF-M or Mbed-TLS repos.

Entire 'psa-arch-tests' suites (Crypto, Initial Attestation and Secure Storage) can be run on TF-M reference implementation. There have been multiple releases for psa-arch-tests since 2019, and TF-M provides mature open source implementation of PSA APIs to run our test suite.

Regards,
Avi.