ARM-software/psa-arch-tests #386: The commands id don't follow the spec.

AllenLius, 2024-09-20:
Ids in spec: DEN0101_Authenticated_Debug_Access_Control_1.0.1.pdf @ p35:

| ID | Constant | Command | Description |
| --- | --- | --- | --- |
| 0x0001 | ADAC_DISCOVERY_CMD | Discovery Query target properties | Discovery Query target properties |
| 0x0002 | ADAC_AUTH_START_CMD | Authentication Start | Initiate authentication sequence; receive challenge |
| 0x0003 | ADAC_AUTH_RESPONSE_CMD | Authentication Response | Send authentication data |
| 0x0004 | ADAC_CLOSE_SESSION_CMD | Close Session | Terminate command session |
| 0x0005 | ADAC_LOCK_DEBUG_CMD | Lock Debug | Lock debug access; restore system to locked state |
| 0x0100 | ADAC_LCS_CHANGE | Change life-cycle state | Change the life-cycle state of the target |

Ids used by the test suite:
```c
typedef enum {
    ADAC_DISCOVERY_CMD = 0x01, 
    ADAC_AUTH_START_CMD = 0x02,
    ADAC_AUTH_RESPONSE_CMD = 0x03,
    ADAC_RESUME_BOOT_CMD = 0x05, // shall be 0x0004
    ADAC_LOCK_DEBUG_CMD = 0x06,  // shall be 0x0005
    ADAC_LCS_CHANGE_CMD = 0x07,  // shall be 0x0100
} adac_commands_t;
```

avinaw01-arm, 2024-09-23:
Hello @AllenLius,
Thanks for spotting this issue. May I know which partner company of the repository you're representing? We need to maintain an internal list of all our issues and partners, that's why asking.

Thanks & Regards,
Avi.

avinaw01-arm, 2024-10-08:
Hello @AllenLius,
Humble request you to respond to our previous message.
In addition, we would like to inform you that after investigating the issue raised, we discovered that these enumerations are defined in our reference platform implementation. We have already reached out to the team responsible for maintaining that repo. Since this fix is not under the Compliance team's scope, we'll need to wait for the relevant team to address it. Thank you for identifying the issue. I’ll make sure to keep you in the loop regarding any updates related to this matter.

Regards,
Avi.

AllenLius, 2024-10-15:
> repository

Hi Avi,

The company is Analog Devices.

avinaw01-arm, 2024-10-15:
Hi @AllenLius,
Noted, thanks for your response.
The other team has started their work on the mentioned issue. I'll get back to you once it's fixed on their end.

Regards,
Avi.