ARM-software/psa-arch-tests #399: Test c_061/63 causes a buffer overflow and a stack overflow

waleed-elmelegy-arm, 2025-03-03:
In test test_c061.c in this call:
```
        /* Finish encrypting a message in an AEAD operation */
        status = val->crypto_function(VAL_CRYPTO_AEAD_FINISH, &operation, output + length,
                 check1[i].output_size, &finish_length, tag, check1[i].tag_size, &tag_length);

```
and in test_c063.c in this call:

```
    /* Finish authenticating and decrypting a message in an AEAD operation */
        status = val->crypto_function(VAL_CRYPTO_AEAD_VERIFY, &operation, output + length,
                 check1[i].output_size, &verify_length, check1[i].tag, check1[i].tag_length);

```
output buffer starts from output+length but size parameter is passed as check1[i].output_size which can be `BUFFER_SIZE` which means we can overflow the output buffer by `length` bytes which also causes a stack overflow.

jk-arm, 2025-04-02:
@shrutig-arm if we reached the mid point then could you please update the state?

shrutig-arm, 2025-04-20:
This issue has been resolved with the [PR](https://github.com/ARM-software/psa-arch-tests/pull/401) .

closed 2025-04-20 by shrutig-arm